spacetelescope/jwql #134: Integrate coveralls

bourque, 2018-08-29:
Now that our repo is public, we can integrate `coveralls`: https://coveralls.io/

bourque, 2018-09-08:
`codecov` is an alternative to `coveralls` that we may want to consider instead: https://codecov.io/

bourque, 2018-09-18:
`coveralls` needs to store a secret token in our `Jenkinsfile` in order to work, but we should not expose this in our public repo.  It looks like Jenkins has a `credentials()` function that will allow us to get the token (https://jenkins.io/doc/book/pipeline/jenkinsfile/#handling-credentials).

@SaOgaz Do you happen to know how/where the `credentials()` function is grabbing the token from?  I imagine we would have to store the hard copy of the token somewhere on the Jenkins server.

SaOgaz, 2018-09-19:
I'm not sure about this, I've never set up coveralls before.

bourque, 2018-10-31:
Thanks @rendinam, I'm looking into `codecov` as an alternative.  It seems in both cases, these things are trivial to set up for Travis CI and non-trivial to set up with Jenkins, and unfortunately we are using the later.

bourque, 2018-10-31:
I have initialized a `codecov` project here:  https://codecov.io/gh/spacetelescope/jwql.  I tried following the instructions on how to configure the project (#170), but on first pass I can't seem to get it working.

SaOgaz, 2018-11-30:
After much poking around with @rendinam, we failed to get something working for this.  We were hoping we could use environment variables, but this wasn't working initially because of some issues on the Jenkins side.  @rendinam is going to think about this situation more broadly with @jhunkeler to see if they can find any solutions.

PR #206 (2019-03-27): Implement code-cov 
Closes #134
closed 2019-03-27 by bourque