spacetelescope/jwql #135: Consider/Research what monitors we may want to create for our monitors

catherine-martlin, 2018-08-30:
Designing something to track the logging results that may be important as we go - for example, how long do the monitoring scripts take to run over time? This may help us recognize if anything is inefficient or needs to be better organized/run.

catherine-martlin, 2018-08-30:
Can look into also tracking the timing for certain types of actions taken by the scripts - for example, when we query MAST how long does that take? When we perform a specific function how long does that take? How are those timing bits changing?

bourque, 2018-08-30:
To help with this, we may want to consider a standardized way to log things.  For instance, it would be easier to determine how long a MAST API query takes if we make sure to log something before and after the query, e.g.:

```python
logging.info('Begin query')
# Do the query
logging.info('End query')
```

We could do something similar for other common operations (e.g. copy a file, saving an output product, etc.)

bhilbert4, 2018-08-30:
Maybe better to be more specific since there can be queries to lots of different databases.

'''
logging.info('Begin astroquery.mast')
# Do the query
logging.info('End astroquery.mast')
'''

bourque, 2018-08-30:
Excellent point!

bourque, 2018-08-30:
Perhaps we could expand the `logging_functions` to have some canned

bourque, 2018-08-30:
Whoops, sorry, didn't mean to close the issue there.  

Perhaps we could expand the `logging_functions` module to handle some of this.  For example, create a function `log_query()` that takes in a query operation and logs it with a standardized statement.

catherine-martlin, 2018-08-30:
That could be useful - then it'd be easier to build in methods to track how long things take, etc.

catherine-martlin, 2019-01-23:
Checklist of things to build into logging: 
- [] `log_query_time()` to log the time it takes a query to run
- [] `log_function_time()` to log the time it takes a function to run

catherine-martlin, 2019-02-13:
Ideas for other things: FileIO operations? Functions to help profile your monitor while you're coding (is that the function_time option? Are there others?). 

Build a function to pop out and monitor the amount of time a script runs on a given day and plots over time -- the log file already tracks the CPU and real time so creating a generalized script to grab those and plot them could be useful.

catherine-martlin, 2019-03-06:
See #327 for current PR regarding this.

closed 2019-04-08 by catherine-martlin