spacetelescope/jwql #142: Build form to submit anomalies

bourque, 2018-09-11:
We will need a way for users to submit an anomaly that they visually identify in an image.  For WFC3, we did this with a set of checkboxes along side the observation (see below).  However, I am open to any better ideas.

Upon submitting an anomaly, a new record should be entered into our `Anomalies` table in our database (see #79).

[image]

hover2pi, 2018-09-11:
I'll take this ticket if no one else wants it.

bourque, 2018-09-11:
@hover2pi Go for it!

bourque, 2019-03-04:
@hover2pi Have you been able to connect and use the development database?  Is there anything I can do to help integrate this?  It would be nice to have this as part of our next release.

closed 2019-06-05 by bourque