spacetelescope/jwql #144: Create private Jenkins server that has access to central storage

bourque, 2018-09-11:
We have several tests (and presumably many more to come) that require access to the `config.json` file.  However, since we cannot commit that file to the repository, we need a setup wherein Jenkins has access to the file via a central storage location.

bourque, 2018-09-11:
@SaOgaz is already working on this, but I just wanted to formally create a GitHub issue for it.

bourque, 2018-10-22:
This issue is not so much related to our code base, so I am going to move it to Jira.  See https://jira.stsci.edu/browse/JWQL-20

bourque, 2018-10-22:
Testing to see if I can successfully link this to the existing Jira issue.

https://jira.stsci.edu/browse/JWQL-20

SaOgaz, 2019-02-06:
Current status: @rendinam and @jhunkeler were looking to a way to get access to the config file on central store with a private jenkins run. Although the private run can access central store, we want a way to tell jenkins where that file is, without storing the directory location in our repository. This is the current blocker.

closed 2019-03-25 by bourque