spacetelescope/jwql #147: Build script execution status display on dashboard

bourque, 2018-09-17:
The dashboard on the web application should display the success/failure of the various `cron` jobs (e.g. `monitor_filesystem`) so that the JWQL team can easily identify scripts that are failing or have not executed when they should have.

Here is an example from what WFC3/Quicklook uses:

[image]

Each row in the table corresponds to a particular `cron` job.  A user can click on the name of the monitor to display the contents of the corresponding log file.  The color indicates how recently the script executed successfully; bright green means that the script executed successfully that day, light green means the script executed successfully several days ago, red indicates that the script crashed.

bhilbert4, 2018-11-12:
So just to make sure I'm on the right track with this, I need to find the latest log file for a given monitor/script being run in the crontab (by looking into the proper subdirectory of our logging directory), check to see if it indicates success and at what time, and then this can go onto a webpage? Does that make sense, or is there a cron job log file somewhere that I should parse for this?

bourque, 2018-11-13:
@bhilbert4 That's correct.  Instead of worrying about the `crontab`, I would suggest to simply look at and parse what is in the `logs/` directory in our central storage area, as you elude to.  There is no `cron` log file.

bhilbert4, 2018-11-14:
@bourque I've been thinking about display options for this. The light green/dark green seems like it can be a little confusing given that different monitors will run at different cadences. I was thinking of trying something like:

green: latest execution was successful
yellow: based on the cadence of the log files, I expected to see a more recent run of this, but I don't
red: latest execution failed

What do you think? I don't have a good intuition for how HST QL teams used this table, and therefore what information is the most important to display.

bourque, 2018-11-14:
@bhilbert4 I like that idea a lot, and agree the various shades of green is not ideal for the reasons you mention.  I endorse that you implement it this way.

bhilbert4, 2018-11-16:
Also, my current function for parsing a log file looks for 'complete' (e.g. "filesystem monitor complete") in the final line of the log in order to determine if the job finished successfully. If 'complete' is not present in the final line, the job is flagged as failing. I'm sure this can be made a little more robust, but it does seem like it would be helpful to have some minimal logging guidelines for all monitors (e.g. at the end of the monitor, state that the monitor is complete or finished , or finished successfully).

bourque, 2018-11-16:
@bhilbert4 Yes, I believe this should go on the dashboard -- the table is for us (the dev team) to be able to tell what is working and what is failing, and so I would like to keep it separate from where our users might wander to within the web app.  I think eventually when we have instrument-specific monitors, they can all just go into the same table.  Perhaps we will want to add some separate columns or row-dividers at that point, but that is a problem for later.

Also, great point about logging guidelines.  I agree that we should document some, and it would also help us with #135

closed 2018-12-17 by bourque