spacetelescope/jwql #155: Upgrade to newer version of Bokeh

gkanarek, 2018-09-25:
I'm sure this will break some things, but we really need to upgrade to a newer version of Bokeh - at least 0.13, but 1.0 is apparently scheduled to be released soonish.

jason-neal, 2018-10-26:
I have tried installing bokeh 1.0.0 and simply running pytest in the repositoy.
I report what I have found so far:

`bokeh.charts` has been removed but the `Donut` plot is used in
`jwql/utils/monitor_template.py` and `jwql/jwql_monitors/monitor_mast.py`

The documentation mentions that bokeh.charts is just a wrapper for [bkcharts](https://github.com/bokeh/bkcharts) which is a archived (unmaintaned) repo that is not compatible with bokeh 1.0.0. It is suggested that 

> For actively developed, supported, very high level charting on top of Bokeh, see the HoloViews project: http://holoviews.org/

So there may be some equivalent for the `Donut` plot in [holoviews](http://holoviews.org/).
While Bokeh's plotting is now under `bokeh.plotting`.

Also in `jwql/jwql_monitors/monitor_mast.py` the functions `save` and `output_file` now need to be imported from `bokeh.io` instead of `bokeh.charts`

`from bokeh.io import save, output_file`

bourque, 2018-10-26:
Thank you for summarizing @jason-neal. Now that `1.0.0` is released, we will update our code to use it, and look into using `holoviews` where needed.

bourque, 2018-10-26:
FYI, more information about the 1.0 release: https://bokeh.github.io/blog/2018/10/24/release-1-0-0/

bourque, 2018-11-06:
According to [`holoviews` twitter](https://twitter.com/HoloViews/status/1056941517963780096) there is a new release that is compatible with `bokeh` `1.0`, and though they claim that it is available in `conda`, it doesn't appear to be:

```
(jwql) [kramer:jwql_monitors] conda install holoviews=1.10.8
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - holoviews=1.10.8
```

bourque, 2018-11-06:
Turns out it was in the `conda-forge` channel: `conda install -c conda-forge holoviews`

bourque, 2018-12-06:
Ok, we should be good to go with this now that #176 is merged and #202 will soon be merged (which upgrades the web app to use 1.0).

closed 2018-12-06 by bourque