spacetelescope/jwql #159: Update web app db interfaces to use Django models

laurenmarietta, 2018-10-18:
I've learned at DjangoCon that it seems we should be interacting with both the MAST archive and our new anomaly database using Django models.

Someone said this to me:

> If you ever find yourself writing SQL in Django, you’re _probably_ doing something wrong.

Plus, I have learned that it is possible to use Django models with existing databases (e.g. MAST): https://docs.djangoproject.com/en/2.1/howto/legacy-databases/

This would mean updating/removing `database/database_interface.py` and `website/apps/jwql/db.py` to instead implement interfaces with those databases in `website/apps/jwql/models.py`.

laurenmarietta, 2018-10-18:
Beyond being more "djangonic" (this is a word I just made up), using Django models also protects against SQL injection!

laurenmarietta, 2019-01-22:
After investigating this a bit, it seems to me like a good way to go for any interactions we'll have between the anomaly database (or whatever other SQL databases might arise... looking at you, #222) and the web application. However, I could see there being an argument for just continuing to use the ORM developed in SQL Alchemy in #117, if someone thinks that is wiser... A debate!

I don't think it makes sense to use Django models to interface with MAST, since there is already a fully developed API there that works just fine.

Additionally, I'm a little concerned about security after all. Using Django's `inspectdb` function to object-ify an existing database would generally mean committing a Django-ized representation of the entire schema of that database to the repo. If that isn't something we'd want to do, I have a vague idea for how we might avoid it, but it might take some finagling. Something to discuss with @bourque? 

At the moment we aren't interacting with the existing anomaly DB in the web app at all (right?), so I don't think this issue currently prompts a PR. It would 100% affect #142, however.

bourque, 2019-02-28:
As I mentioned in a meeting with @laurenmarietta and @bhilbert4, I am not convinced at this time that moving to `django` models is going to be beneficial and give us features that we couldn't already get with `sqlalchemy`.  From looking at some `django` models tutorials, it seems that both `django` models and `sqlalchemy` can accomplish the same thing in a similar way, the only difference being in method and parameter names.  For example, a `django` model might look like:

```python
class Author(models.Model):
    """Model representing an author."""
    first_name = models.CharField(max_length=100)
    last_name = models.CharField(max_length=100)
    date_of_birth = models.DateField(null=True, blank=True)
    date_of_death = models.DateField('Died', null=True, blank=True)
```

while the `sqlalchemy` implementation would look like:

```python
class Author(Base)
    """ORM for the author table."""
    first_name = Column(String(100))
    last_name = Column(String(100))
    date_of_birth = Column(Date, nullable=True)
    date_of_death = Column(Date, nullable=True, default'Died')
```

Also, since there will be some interactivity with this database outside of the `jwql` web application (i.e. instrument monitors reading from/writing to tables via Python scripts), it may be a bit weird to have the database exist completely under the `website/` umbrella.   

For these reasons, and those mentioned above by @laurenmarietta, I think we should just continue developing using SQLAlchemy for now.  Of course, if things change or we discover useful features of Django Models, we can certainly move towards that in the future!

closed 2019-02-28 by bourque