spacetelescope/jwql #160: Consider using Black?

laurenmarietta, 2018-10-18:
Black is the "uncompromising Python code formatter": https://github.com/ambv/black 

It is certainly uncompromising, so might not be best for our project, but a lot of people here at DjangoCon are really into it for enforcing consistent programming across a project. Something to consider!

bourque, 2019-03-01:
For funsies, I went ahead and ran `black` on the `jwql` package, committed the changes and pushed a branch to my fork of the `jwql` repo.  https://github.com/bourque/jwql/pull/1 will show everything that would change if we adhered to `black`.  Perhaps we can look at this in a future meeting and discuss whether or not we want to adopt any or all of this.

laurenmarietta, 2019-03-01:
😲🏴🖤⚫️

bourque, 2019-05-30:
An enlightening talk from PyCon 2017: https://www.youtube.com/watch?v=esZLCuWs_2Y

bourque, 2019-05-30:
I am going to try to keep a pros & cons list here:

Pros:
- It appears to be gaining popularity within the python community
- It is uncompromising, so we would spend less time and effort  reviewing and changing code based on formatting
- There are equivalent tools in other popular languages (e.g. `go` has `gofmt`, `javascript` has `standardjs`)
- We could employ a git pre-commit hook to ensure commits are blackened 
- 

Cons:
- It is uncompromising, so we are forced to use a very specific standard
-

bourque, 2020-03-25:
I don't see us having the resources to be able to implement and enforce this anytime soon, so I will close for now.

closed 2020-03-25 by bourque