spacetelescope/jwql #161: Ensure web application is accessible!

laurenmarietta, 2018-10-18:
Some tips for making web applications as accessible as possible, for differently-sighted users (and also just making apps better in general):
* Using HTML
    * Provide alt tags for images
    * Pay attention to the tabindex
    * Headers (h1-h6): actually use them! Organize documents like trees, and don’t skip levels. 
    * Mark things as headers, articles, figures, footers. (Look here: https://www.w3schools.com/tags/)
* Using ARIA to actually define the roles of different tags (e.g. role=“button”): https://developer.mozilla.org/en-US/docs/Web/Accessibility/ARIA

Web content disability guidelines: https://www.w3.org/TR/WCAG21/

laurenmarietta, 2018-10-23:
ST also has a page with a LOT of resources on web app accessibility: https://innerspace.stsci.edu/pages/viewpage.action?spaceKey=OPO&title=Accessibility

bourque, 2018-10-23:
Thats cool that this exists!

bourque, 2019-06-05:
Per @catherine-martlin suggestion, maybe we can tackle this in a few hours long hack session in a conference room!

PR #457 (2019-07-16): Accessibility Guidelines
This PR includes various changes to the web app templates to conform to accessibility guidelines (https://innerspace.stsci.edu/pages/viewpage.action?spaceKey=A5C&title=Webpage+Checklist)

Closes #161
closed 2019-07-16 by bourque