spacetelescope/jwql #171: Implement logging for the web app

laurenmarietta, 2018-10-31:
Integrate our existing logging module into the web app, to document what goes on. Not too sure of the specifics for how we would best implement this, but: 

https://www.owasp.org/index.php/Top_10-2017_A10-Insufficient_Logging%26Monitoring

laurenmarietta, 2018-10-31:
Apparently ITSD also supports the use of Splunk, a log management software: https://www.splunk.com/en_us/solutions/solution-areas/log-management.html

stscijgbot-jwql, 2024-05-13:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-241?focusedCommentId=698832#comment-698832):_

Most likely, we may have to go through ITSD to get info? I'm not sure we have read access to all of the various logs on the jwql servers.