spacetelescope/jwql #180: Add project description and higher level documentation to ReadTheDocs

bourque, 2018-11-08:
We have a working ReadTheDocs page with API documentation here: https://jwql.readthedocs.io/en/latest/

It would be nice to add some higher-level documentation to go along with this, such as a project description and how to install/use the `jwql` package.

laurenmarietta, 2018-11-08:
Scrolling through the ReadTheDocs, it also seems like we could use another level of headings within the different sub-packages? For instance, under the "utils" tab, the docs for `utils.logging_functions`, `utils.permissions`, etc. all run together. I imagine we would prefer to have headings for those modules  
 (`logging_functions`, `permissions`, ...) within the sub-package docs to delineate between them.

bourque, 2018-11-08:
@laurenmarietta Good point, agreed!

PR #194 (2018-11-26): API Documentation Updates
This PR makes several changes to the API docs, namely:

1. Adds headings before the individual module (addresses @laurenmarietta's suggestion in #180)
2. Adds some project description to ReadTheDocs home page (addresses #180)
3. Some formatting changes to make things look nicer
4. Added section on Issue Reporting/Feature Requests to `README.md`
5. Added some conditional imports to some modules to get around ReadTheDocs build issues (addresses #154).

Closes #154 
Closes #180
closed 2018-11-26 by bourque