spacetelescope/jwql #189: Establish logging guidelines

bourque, 2018-11-16:
Per @bhilbert4 's idea, we should establish some guidelines for how to properly log the execution of a script so that we can build tools that can interact with the logs in a standardized way.  We can document these guidelines in a wiki page perhaps.  This would also help solve some of the ideas in #135.

catherine-martlin, 2018-11-27:
Would it be better to add a section to an existing Wiki such as [Style Guide](https://github.com/spacetelescope/jwql/wiki/Style-Guide) or the original [logging page](https://github.com/spacetelescope/jwql/wiki/JWQL-Logging-Function:-The-Why's-and-The-How)? 

I'm considering an addition to the logging page with a link from the style guide to remind people.

catherine-martlin, 2019-01-23:
A small update was placed [here](https://github.com/spacetelescope/jwql/wiki/Style-Guide) regarding the soon to come `jwql`-specific logging standards.

catherine-martlin, 2019-01-24:
This will be addressed by PR #258

bourque, 2019-04-08:
See #258

closed 2019-04-08 by bourque