spacetelescope/jwql #200: Explore using Docker

bourque, 2018-11-29:
I wonder if we could use Docker to make the packaging and installation of `jwql` easier for users

laurenmarietta, 2018-11-29:
Here is an instructive talk from DjangoCon explaining the basics of Docker - could be useful:
https://www.youtube.com/watch?v=qsEfVSTZO9Q

bourque, 2019-01-09:
Thanks @laurenmarietta, that was a very informative! I'm going to try my hand at implementing something.

laurenmarietta, 2019-02-06:
Potentially useful O'Reilly resource about Docker:
https://learning.oreilly.com/videos/learning-docker/9781491956885/9781491956885-video244778

bourque, 2019-02-14:
I found this resource to also be quite helpful: https://semaphoreci.com/community/tutorials/dockerizing-a-python-django-web-application

JeffValenti, 2019-02-25:
FYI, ITSD did not allow me to install Docker on my JWST-funded laptop, due to security concerns (additional ports need to be enabled). Instead they created a VM with Docker for me to use. Not sure why that was deemed OK. In order to use Docker commands, ITSD gave me sudo privileges for the docker command. Ugly. Getting data in and out of Docker <--> VM <--> local machine required lots of typing and frequent 2FA. Not fun. ITSD is pondering whether there is a better approach. Probably my workflow could be improved as well. If the user interface is only via a web application, then perhaps the use of docker would be transparent to the user. However, at this point my naive impression is that conda is much simpler and hence better than docker, as long as conda provides the functionality required. For now I have abandoned Docker for my personal research project.

bourque, 2019-02-25:
Thanks @JeffValenti, that is good to know.  When talking with @alexyermolaev who was helping to set up our development web server, he had mentioned that the process would have been "easier if I had a docker container", and so that prompted me to go down that road.  I have a primitive working example (#297) that may or may not be used to help us set up a production web server (#294), but that will be up to ITSD I suppose on whether or not they want to use it.  This is all to say -- hopefully ITSD will give some further guidance on if and how these docker containers can be used effectively for this use case.

bourque, 2019-04-29:
As we move towards a solution that uses ASB infrastructure (#294), it appears that Docker is not going to be needed, so I am closing this issue for now.

closed 2019-04-29 by bourque