spacetelescope/jwql #210: Display which version of jwql is being used in the web app

bourque, 2018-12-04:
A cool idea inspired by the `exoctk` project -- We could put a small piece of text in the corner of the web app somewhere that indicates which version of `jwql` is being served.

frboffi, 2018-12-04:
yes! this is important to have!

bhilbert4, 2019-02-11:
What's the easiest way to find the version being run? Read it from `setup.py`? It looks like we don't currently have `__version__` set to anything.

bourque, 2019-02-11:
@bhilbert4 Yeah, I would say normally the best way is to do 

```python
jwql.__version__
```

but we don't have it set up.  I've never actually looked into doing that before.  Maybe you could include that as part of this PR?  I'm not sure what one has to do to provide this, but I imagine someone on StackOverflow has done this before.

laurenmarietta, 2019-02-11:
We could also add a version badge to the README:
https://img.shields.io/github/release/spacetelescope/jwql.svg

bhilbert4, 2019-02-11:
Lots of options: https://packaging.python.org/guides/single-sourcing-package-version/
Seems like reading from `setup.py` might be easiest. Most of the other options involve defining the version number somewhere other than in `setup.py`. Do you see any options in there that are more appealing?

For a moment I was excited about option 5 (using `pkg_resources.get_distribution`), but since that only knows about the installation metadata, it returned a very old version number for me, as I installed jwql a long time ago and I don't re-run `setup.py` for each new version.

bhilbert4, 2019-02-11:
But perhaps one of these options can make it such that the version number in `jwql` is stored in only a single place, rather than in `setup.py` and `conf.py`?

bourque, 2019-02-11:
@bhilbert4 Would it be possible to read in the `VERSION` in `setup.py` from the `jwql.__init__.py` file?  Something like:

```python
with open('../setup.py') as f:
    data = f.readlines()
__version__ == #parse out VERSION
```

Even if we had a separate file that stored the version, I think we would still need a similar code in `jwql.__init__.py` to define the `__version__` variable, unless I am misunderstanding.

bhilbert4, 2019-02-12:
Yes, that sounds right. It seems like common practice is to read in the version from somewhere (setup.py, __init__.py, etc). I'm wondering if we read in the version number from setup.py, that we could then also add the code to conf.py to read the version number in from setup.py.

bourque, 2019-02-12:
@bhilbert4 Yeah, that all sounds good.  I don't see why you couldn't do that, though it may involve some `sys.path()` trickery in `conf.py`, since that module is not in the package.

closed 2019-02-18 by bourque