spacetelescope/jwql #215: Explore using relic

bourque, 2018-12-14:
[relic](https://github.com/spacetelescope/relic) is a version management package built and maintained by folks here at ST.  It could help us avoid having to hard code version numbers at various places within our package.

bourque, 2018-12-14:
@Johannes-Sahlmann Do you have a working example of a project that uses `relic` that I could see?  I find that the existing documentation for `relic` is not very informative and its not clear exactly what I would have to do to set it up.

bhilbert4, 2018-12-14:
Both `pysiaf` and `mirage` use Relic.
https://github.com/spacetelescope/mirage
https://github.com/spacetelescope/pysiaf

bourque, 2018-12-14:
Thanks @bhilbert4, and now that I think of it, I believe you've already told me this before. Sorry for forgetting!

bourque, 2018-12-18:
I think I understand how `relic` is installed and used now.  Some thoughts I have about it:

- It is pretty easy to use, you just have to stick that snippet of code into the `setup.py` file
- I don't like how it requires you to download a clone of the `relic` package locally
- Using this would not offer an immediate solution to providing the `VERSION` in `docs/source/conf.py`, unless we also put the snippet of code in there
- The version number that `relic` extracts is based on the user's local clone of the repo, so if they have not pulled the latest `tag`, the version number would actually be outdated.

Perhaps an easier solution is something like this:

```python
In [1]: from bs4 import BeautifulSoup
In [2]: import urllib.request
In [3]: url = urllib.request.urlopen('https://github.com/spacetelescope/jwql/releases.atom')
In [4]: soup = BeautifulSoup(url)
In [5]: tag = str(soup.find('entry').find('id'))
In [6]: version = tag.split('</id>')[-2].split('/')[-1]
In [7]: print(version)
0.16.0
```

bourque, 2019-01-07:
@bhilbert4 Yes, except that `releases.atom` doesn't need to be manually updated.  The version number that is contained in that file is just automatically pulled from whatever the latest `tag` is on GitHub.

bourque, 2019-01-17:
It does seem that `pysaif` has a workaround for getting the proper version number into `docs.conf.py` (fyi @Johannes-Sahlmann):

```
package = importlib.import_module(setup_cfg['package_name'])
version = package.__version__.split('-', 1)[0]
# The full version, including alpha/beta/rc tags.
release = package.__version__
```

However, I think I would like to try my solution, as least for a while. If it turns out to be buggy or unreliable, we will explore using `relic` again.

bourque, 2019-01-23:
Turns out that my solution (#250) does not work really well, especially in the case where a user is trying to install a previous version of the code, as @Johannes-Sahlmann points out.  

Perhaps we should kick the can on using `relic` again.

bourque, 2019-02-28:
I am going to close this.  I think the workflow we are currently using is working just fine for now, but if that changes, we can re-explore this.

closed 2019-02-28 by bourque