spacetelescope/jwql #218: Develop software for Dark Current Monitor

bourque, 2018-12-14:
Design, develop, and integrate software for the dark current monitor.  For context not related to the software, see Jira issue: https://jira.stsci.edu/browse/JWQL-3

bhilbert4, 2018-12-15:
As I was thinking about high level design this afternoon, I was thinking about how this (and other) monitors would be run. 

Nominally the monitors will be run by cron job, right? But really what we want to do for a particular monitor is to run it each time a new, appropriate dataset shows up in the archive. So would it make more sense to run a cron job that queries the archive/filesystem, and if new files are present, the actual monitor is kicked off?

bhilbert4, 2018-12-17:
Nevermind. Assuming that the dark current monitor wraps around a file search function as described above, then it makes sense to run the dark_current_mointor by cron job on some regular schedule. That's what I get for thinking about this late on a Friday night.

bourque, 2018-12-17:
@bhilbert4 Yes, I imagine there will be a file search function of sorts in each monitor, that looks for new data.  However, I think you bring up a good point -- it would be nice to have the monitor update as soon as new data arrives.  

One thing that we could do is have a button on the web app -- "Update monitor" or something -- that spawns a new run of the given monitor.  The user could then be brought to a webpage that indicates that a new run has been executed and to check back in the future for an updated plot.  In this way, if a monitor is outdated because new data has arrived since the last cron run, at least a user can manually update things.

laurenmarietta, 2018-12-17:
I'm not sure if this is applicable here, but I know that the OTE commissioning procedures for how image analysis is done in the wavefront office includes something called a DAN (Data Availability Notification).  One machine receives a DAN when a user on another machine delivers files to a certain directory.

In an ideal world, the JWQL server would receive a DAN from MAST every time new data is available, and that could trigger monitor runs. I have no intuition for what it would take for this to be implemented, though.

JeffValenti, 2018-12-17:
JWQL should subscribe to the MAST [notification service](https://jwst-docs.stsci.edu/display/JDAT/Data+Discovery?q=notification%20service#DataDiscovery-Anticipateddata), if you want MAST to tell you when new data are available in the archive. Ask Caty Kaleida for more information. DANs are for communication between S&OC subsystems and are not available to "external" users.

laurenmarietta, 2019-05-29:
Should this issue be closed, since #229 was completed?

bourque, 2019-05-29:
Yep! I would say so!

closed 2019-05-29 by bourque