spacetelescope/jwql #219: Develop software for Target Acquisition Monitor

bourque, 2018-12-14:
Design, develop, and integrate software for the Target Acquisition monitor.  For context not related to the software, see Jira issue: https://jira.stsci.edu/browse/JWQL-23

bourque, 2019-03-13:
@gkanarek I'm thinking about putting this into our next sprint.  Do you think you would have time to start working on this in the next month?

bourque, 2019-03-18:
@cracraft I am going to assign this to you as well, since it is listed as a high priority for the MIRI team.  Do you know if any current work is being done for this for the MIRI team, or if there is some already existing software?

cracraft, 2019-03-20:
I have sent an email to the team asking for any scripts, current work, etc, and am waiting for any responses.