spacetelescope/jwql #222: Integrate Bokeh into the Django backend

gkanarek, 2018-12-14:
The current monitor scripts serve a dual purpose: both that of actually performing the relevant monitor task, and also of creating the html and Javascript output to be embedded in the relevant webpages, as required. I propose a new architecture, where the Bokeh code is integrated with the Django backend, and reducing the monitor scripts to standalone monitoring tasks. This would improve on the existing setup in a few ways:

1. It will be easier for users to make use of the monitor scripts on their own for various analysis tasks, rather than having to deal with Bokeh output.
2. Making the Bokeh code part of the Django backend will allow us to leverage the power of Bokeh more effectively (e.g., streaming data to plots, not having to re-create html code every time a monitor script is run, etc).
3. People who develop and maintain the monitor scripts don't have to be versed in Bokeh or Django.

Before we can make this change, we need to decide on a standardized methodology for storing the results of the various monitors which can be read by the Bokeh code. Then, each script will need to be split in two: the actual monitoring task (which will be run via a cron job and store its output in the standardized format), and the web interface, which will be incorporated into the Django backend.

bourque, 2019-03-04:
FYI, We have come up with a generic way to store instrument monitor data (#304) in the form of database tables.

bourque, 2019-04-18:
@gkanarek Could you open a WIP pull request for your work on this?  It will help us reference and coordinate the work in other issues and pull requests.

bhilbert4, 2024-05-13:
We have moved away from Bokeh templating.

closed 2024-05-13 by bhilbert4