spacetelescope/jwql #226: Require asdf >= 2.3.0

laurenmarietta, 2018-12-17:
I just tried to run the test suite, and got a failure from `test_preview_image.py`: 

```
ModuleNotFoundError: No module named 'asdf.types'
```

It looks like the `asdf.types` module is new to `asdf` 2.3.0, which isn't specified in our `environment.yml` file. I recommend we add this specification, rather than revert the code in the test, since the [`asdf` docs](https://asdf.readthedocs.io/en/latest/asdf/changes.html) say they intend to completely do away with the old syntax (`asdf.asdftypes`) with version 3.0.

(I am concerned that this problem wasn't caught by Jenkins...)

bourque, 2018-12-28:
@laurenmarietta I think that Jenkins did not pick up on this error because it was not actually testing any files.  `test_preview_image` requires there to be a `test_data/` directory, and the files within it will be the ones that will be tested.  I don't think we have set this directory up on the Jenkins server.

Now that I think about this further, I'm not sure how it would be possible to set up a `test_data/` directory with files in them on our existing Jenkins server (perhaps @SaOgaz knows a way around this), unless we commit such a directory + files in our repository.  And seeing as though FITS files are huge and could really cause issues with our repository, I don't really want to go down that road.

The best path forward could be that once we get a private Jenkins server set up and working (#144), we could just have `test_preview_image` point to some files located in central storage. 

For the time being, I think we will just have to rely on local tests of `test_preview_image` to discover any issues with `preview_image.py` and `generate_preview_image.py`.

SaOgaz, 2018-12-30:
I don't think there's a good work around besides setting something up for our private run.  We could also have the private Jenkins run copy the files over with a shell script...

laurenmarietta, 2019-01-24:
Closed via #238.