spacetelescope/jwql #231: Tracking resources and schedule for JWQL in 2019

frboffi, 2018-12-19:
We need to track JWQL development in 2019.
Plan of action:
1) identify development areas based on what we know - Francesca and Matthew
2) drafting a timeline and FTE estimates - Francesca, Matthew, with each of the main contributors for each area - Francesca scheduling meetings in January
3) draft a project schedule with deliverables
4) review plan with JWST mission office
5) once we have finalized with mission, we will share with INS MT to firm up resources
addendum: introduction to project management (waterfall and agile) will be given in January - Francesca and Matthew

bourque, 2019-01-22:
Some take-aways from our meeting with Joe Zahn:

- We could assign a PERT number to each item in our product backlog, this would help in sprint planning and scheduling for FY19
- We could calculate the capacity that each development team member has for FY19, broken down per year/month/week/day

frboffi, 2019-01-23:
For everyone's reference, might be useful to reference:
https://www.linkedin.com/pulse/what-pert-how-can-we-use-dave-fourie-pmp-prince2-
(at least review what the formula is and what the terms are).

frboffi, 2019-01-24:
Providing more specifics about activity duration estimation techniques:  https://innerspace.stsci.edu/display/JWQLPROJ/Estimating+Activity+Duration

closed 2020-03-25 by bourque