spacetelescope/jwql #232: Expand tests for api_views

bourque, 2018-12-19:
As @frboffi and @catherine-martlin suggested, it would be a good idea to expand the tests in `test_api_views` to include each of the five JWST instruments.  Currently, we just have tests for various data from FGS.

laurenmarietta, 2018-12-22:
Heads up that this might end up being a bit tricky - it looks like every instrument except FGS has a file in the filesystem with funky formatting that is causing `filename_parser` to throw an error. Some examples:
- jw00733001001_02101_00004-**seg001_nis**_uncal.fits
- jw00733003001_02101_00002-**seg001_nrs1**_rate.fits
- jw94015001002_02104_00001_nrcblong_**o001**_crf.fits
- jw**80600-o009_t001**\_miri\_**f1130w**_i2d.fits (!!?!)

bourque, 2018-12-23:
@laurenmarietta Thanks for digging up those examples.  I think I will have to ask some of the archive folks why there are files in the filesystem that don't conform to the naming convention.  Perhaps the copy of the filesystem that we have been using is now outdated and the issue has since been fixed.

rizeladiaz, 2018-12-24:
Those seg001 files are Time series observations. The TS observations can be very large, so these will be segmented in files of < 2Gb so these can be calibrated and distributed.
The last example is the product of an association created at Stage 3 of the calibration pipeline.

rizeladiaz, 2018-12-26:
The official sources of information (latter to be merged) are:

https://jwst-docs.stsci.edu/display/JDAT/File+Naming+Conventions+and+Data+Products
and
https://jwst-pipeline.readthedocs.io/en/stable/jwst/introduction.html#output-file-and-associations

bourque, 2018-12-28:
Thanks for the information @rizeladiaz! It is quite helpful!

PR #281 (2019-02-07): More API views tests
This PR adds several tests to `test_api_views` in order to test more instruments, proposals, and filenames.  It also refactors a few of the functions in `data_containers` in order to speed them up, as I was finding some of the views to be really slow when dealing with a large amount of data.

Closes #232
closed 2019-02-07 by bourque