spacetelescope/jwql #254: Troubleshoot web app loading times

laurenmarietta, 2019-01-22:
The fact that pages on the web app can load really slowly has been a longstanding issue that we should address sooner rather than later.

While on the VPN from my house, I have been waiting 10 (!) minutes for one MIRI proposal worth of thumbnails to load. These times are prohibitive for any future users, and I'm sure they're because I did something wrong when I first set the web app up.

Maybe the Django debug toolbar will be able to help?

bourque, 2019-01-22:
@laurenmarietta I definitely agree that the django debug toolbar could be useful for this.  FYI, I attempted to get it installed, but can't seem to get it working (#249).  You might have better luck.

My hunch is that the hangup has something to do with iterating through the filesystem to grab the data to display.  If that is indeed the case, perhaps we can strive to get appropriate data from an `astroquery.mast` query instead.  Generally speaking, database queries are faster than file searches, especially over a network.

laurenmarietta, 2019-04-15:
Closed via #348.

closed 2019-04-15 by laurenmarietta