spacetelescope/jwql #257: Ask IT for test database

bourque, 2019-01-23:
We could use a test `postgresql` database that we can use as a sandbox for development purposes.  This would help the likes of @hover2pi with #142 and @SaOgaz and @cracraft with #86.

SaOgaz, 2019-01-23:
From what I can gather talking to IT about the reset problem, we should make our current DB the test one, and ask for a permanent production DB.

bourque, 2019-01-23:
@SaOgaz Good point.  In that case we can ask IT to set us up with a database on one of our production servers and then use the one on the dev server as a sandbox

bourque, 2019-01-31:
I have requested a production database from IT!

bourque, 2019-02-04:
We now have a production database! I will put instructions in the home directory of the production VMs with instructions on how to access.

closed 2019-02-04 by bourque