spacetelescope/jwql #263: Add engineering database page to web app

Johannes-Sahlmann, 2019-01-28:
The jwql web app should have a page that allows the user to interface the DMS EDB for simple (single) mnemonic query and graphic display.

More advanced functionality would be a search bar that allows to explore the EDB inventory by name (wildcard syntax), description, category, etc.
That functionality will rely on caching the mnemomic inventory in regular intervals, instead of retrieving it any time the web app interface is used.

Johannes-Sahlmann, 2019-01-29:
@bourque @laurenmarietta Do you have experience with caching in python? 
Here I would like to cache the inventory of available mnemonics (I plan to store that in an astropy table with ~15000 rows and <10 columns). I came across functools (https://docs.python.org/3/library/functools.html#functools.lru_cache) which looks promising. 
The functionality that I would like to achieve is that the web-app refreshes (i.e. downloads) the mnemonic inventory maybe once a day at most.

laurenmarietta, 2019-01-29:
Unfortunately no, I have never worked with caching before. 

It does look like Django has some built-in caching features, though, that might be worth exploring: https://docs.djangoproject.com/en/2.1/topics/cache/
There are a lot of options and the page is quite long, but maybe something relatively simple like database caching would do the trick? I could see using Django's caching with the REST API to add a new mnemonic inventory page to a cache.

Not sure what might be the benefits of caching with Django versus within the Python backend as with `functools`.

bourque, 2019-02-07:
This is technically completed via #272.  We can open separate issues and PRs for changes and improvements moving forward.

closed 2019-02-07 by bourque