spacetelescope/jwql #279: Determine how best to store monitoring data

bourque, 2019-02-05:
Now that we have a few monitors in the works (#229 , #269), we should decide on a generalized/standard way to store monitoring data.  This could take the form of a generalized class or module with convenience functions for saving, reading, displaying data.

bhilbert4, 2019-02-06:
Actually there are a couple images that would be worth storing for the dark monitor: The "baseline" or comparison mean dark slope image and stdev image. These would be needed for each instrument/detector/aperture combination.

bhilbert4, 2019-02-26:
This afternoon I started thinking about database columns for the various tables to use for the dark monitor. This brought up a question: Are we going to try to store data for potential more in-depth manual analyses by users, or should we say that in-depth analyses are not the job of JWQL?

For example, I was thinking about dark current histograms. If a user sees an odd looking histogram that they want to investigate more, will they need to repeat the MAST query, run the pipeline, create a mean slope image, and re-calculate the histogram? Or should we give them the histogram (via a set of x, y values, or some kind of bokeh object)? Or should we do something in between, such as give them the mean dark current image and let them generate the histograms?

bourque, 2019-02-27:
@bhilbert4 I think the responsibility of JWQL here is to execute the monitor, provide any output data that it produces (whether it be a plot, data points, images, database tables, etc) _within the scope of the monitor_, and provide it to the users in a way that it easy for them to read from or make a copy of.  By "within the scope of the monitor", I mean that we only have a responsibility to provide data that is produced by the software of that monitor and nothing else -- If a user is unsatisfied with the monitor and wishes to produce something that is different or beyond what we provide, it would be up to them to do it. 

Specifically for the dark monitor, I would say it is up to us to provide the mean slope images, mean dark current images, the histogram plot and the data points used to produce it.  

Do you foresee any issues with this?

PR #304 (2019-02-28): A generic way to store instrument monitoring data
This PR includes supporting software to store instrument-specific monitoring data.  Most definitely a work in progress. 

Closes #279
closed 2019-02-28 by bourque