spacetelescope/jwql #282: Make template json.config file publicly available

Johannes-Sahlmann, 2019-02-06:
There seems to be no easy way to know the required structure and minimum content of the json.config  file. This should be provided somehwere on the public-facing side of jwql, e.g.:
```
The json.config file has be of the following form with the specified keys. 
{
    "filesystem" : "",
    "preview_image_filesystem" : "",
    "thumbnail_filesystem" : "",
    "outputs" : "",
    "jwql_dir" : "",
    "admin_account" : "",
    "log_dir" : "",
    "test_dir" : "",
    "test_data" : "",
    "setup_file" : "",
    "mast_token" : "",
    "client_id" : "",
    "client_secret" : "",
    "auth_mast" : ""
}
```
Relates to #274 #275

bourque, 2019-02-06:
Perhaps the easiest solution is to just commit a blank `config.json` file template to the repository

Johannes-Sahlmann, 2019-02-06:
Yes, that would work. As part of the contributor and reviewer checklist we should then add to verify that the file was updated if a key was added.

bourque, 2019-02-06:
My only concern is that we should be careful to also put it in the `.gitignore` so that changes to this file do not appear to `git` as commit-able.  This would help avoid having someone accidentally committing a `config.json` file with data in it.

bourque, 2019-02-08:
I don't think simply committing a blank `config.json` file to the repository is going to work.  After committing a template, any further changes to the file would make `git` recognize the changes and give you the option to add and commit it, despite the file being listed in `.gitignore`:

```
(jwql) [kramer:utils] pwd
/user/bourque/repositories/jwql/jwql/utils

(jwql) [kramer:utils] ls
__init__.py		constants.py		logging_functions.py	permissions.py		preview_image.py
config.json		engineering_database.py	monitor_template.py	plotting.py		utils.py

(jwql) [kramer:utils] git status
On branch config-file-template
nothing to commit, working tree clean

(jwql) [kramer:utils] git add --force config.json

(jwql) [kramer:utils] git commit -m 'Initial commit of config file template'
[config-file-template 7bd15c3] Initial commit of config file template
 1 file changed, 17 insertions(+)
 create mode 100644 jwql/utils/config.json

(jwql) [kramer:utils] git status
On branch config-file-template
nothing to commit, working tree clean

(jwql) [kramer:utils] echo "some change" > config.json

(jwql) [kramer:utils] git status
On branch config-file-template
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   config.json

no changes added to commit (use "git add" and/or "git commit -a")

(jwql) [kramer:utils] more ../../.gitignore
*.pyc
*.fits
*.ipynb_checkpoints/
/jwql/utils/config.json
build/
dist/
jwql.egg-info/
docs/build/
docs/source/api/
.cache
*.pytest_cache
htmlcov/
```

  After some [digging on StackOverflow](https://stackoverflow.com/questions/9794931/keep-file-in-a-git-repo-but-dont-track-changes), there does seem to be a solution to this:

```
git update-index --assume-unchanged FILE_NAME
```

but it is on a per-repo basis.  This means that each one of us would have to run this command.  While I certainly trust our dev team to do this, I would like to avoid having for our potential users and external contributors to have to do this.  Failure to do so could lead to an easy mistake of accidentally committing a filled-out `config.json` file, which we should try to avoid as much as possible.

Another possible solution is to commit a `config.json` template to the parent directory of the repo, not ignore that file, and tell users that they must copy that file over to `utils/`, where it will be ignored.

FYI @Johannes-Sahlmann.  Perhaps you have some thoughts about this.

bhilbert4, 2019-02-08:
What if you just called the blank config file something like `template_json.config`, or `blank_json.config`. That way it would be more clear what the purpose of the file is. And with a name that's not just `json.config` it makes it less likely that someone would accidentally commit their actual `json.config` file.

bourque, 2019-02-08:
@bhilbert4 Good idea.  I agree that naming it something other than just `config.json` is an added layer that would help prevent bad things.

I think no matter what, having a template `config.json` file somewhere in the repository is a bit of a risk.  All someone would have to do is think that the template file is the actual file that they have to fill out, and add/commit it.  I don't think that is too far out of the realm of possibility for a novice user.  On the other hand, making the user to go out of their way to physically create the file and stick it in the right place gets the user thinking more about the whole process, thus hopefully avoiding any oversight.

I can see pros and cons for both ways and I'm not really sure what the best solution is 🤔

Johannes-Sahlmann, 2019-02-08:
Instead of committing to the repo, we could make the `blank_json.config` file available via the wiki. That may make maintenance a bit more tedious but should be OK.

bourque, 2019-02-08:
@Johannes-Sahlmann Yeah, I think that is the best solution for now.  I'll go ahead and do that.

bourque, 2019-02-08:
Hmm, perhaps we could write something in `setup.py` that creates a template for you and saves it in the appropriate location.

bourque, 2019-02-08:
Updated wiki page here: https://github.com/spacetelescope/jwql/wiki/Config-file

I will update the `README` file to make mention of this.

PR #287 (2019-02-11): Config file instructions
This PR adds a section to the `README` that describes how to provide the `config.json` file when installing the `jwql` package.  This references the recently modified [wiki page](https://github.com/spacetelescope/jwql/wiki/Config-file), which now contains a template file.

Closes #282
closed 2019-02-11 by bourque