spacetelescope/jwql #294: Build production web server

bourque, 2019-02-13:
Build a web server, similar to our development web server, for production purposes.

bourque, 2019-03-14:
@laurenmarietta and I chatted with @mfox22 today about the infrastructure that exists in ASB.  We should have a look, and possibly use some of it for our web application: https://grit.stsci.edu/mast/asb-core/blob/master/doc/README.md

bourque, 2019-04-12:
I reached out to Christian Mesh about this.  We are going to meet sometime and try to hack together a working example.

mfox22, 2019-04-16:
I'd like to sync up with you all too. Let's discuss strategies for:
- Sharing code (automation, test frameworks, Ansible plays)
- Determining if there are opportunities to share infrastructure

bourque, 2019-06-19:
Some notes from meeting with Christian:

Jenkins Build:
push to pypi
generate custom conda file from template
artifact generate conda file
automatically run delpoy

Jenkins Deploy:
param env (dev/test/ops)
conda file that links to artifcat 

Ansible playbook (or just use ssh for now):
On each machine:
- stop service
- delete conda environment *
- create conda env
- start service

(Could put in HOME directory, but better to use "Host setup" w/ 50 GB storage inside of /opt/jwql/ for apps & logs)
cron @reboot command to start service

Also need to svc acct setup ssh into host as svc account from Jenkins

Look at common-mast for a simpler example

bourque, 2020-03-31:
After a meeting with the JWST mission office, we determined that we should have a conversation again with someone in ASB to explore an ASB-powered solution for this.

bourque, 2021-05-24:
I have sent an email to ITSD to ask for a production web server!

bourque, 2021-09-02:
The production web server has been built!  However, there are a few more things to take care of:

- [ ] Implement SSO layer / AD group authentication
- [ ] Change domain name to jwql.stsci.edu

bourque, 2021-10-07:
This is now complete!