spacetelescope/jwql #302: Add download button(s) to EDB web page

Johannes-Sahlmann, 2019-02-21:
Users may want to download the results of their manual exploration of the EDB mnemonics. Appropriate functionality for download should be added.

Johannes-Sahlmann, 2019-02-21:
@laurenmarietta @bourque do we already have a download button implemented in one of our pages? If not, can you point me to useful/applicable resources or examples?

Johannes-Sahlmann, 2019-02-21:
This may be relevant: https://docs.djangoproject.com/en/2.1/topics/files/

laurenmarietta, 2019-02-21:
We do, yes. Check out the `view_image` template/view; there are buttons there that let users download FITS/JPEG files of the image they are currently viewing.

What I anticipate will be complicated is that Django needs to be able to access the files for download. To download the FITS/JPEG image files, we added the `jwql_dir` value from the config file to the `STATICFILES_DIRS` value in `jwql_proj/settings.py`. If you want to download files from somewhere else, though, we might need to figure out a different way to facilitate that.

laurenmarietta, 2019-02-21:
Ah, the Django link you provided might be helpful with the complicated bit! I can't quite tell how their `File` object would be passed to a template to be download-able, though.

Johannes-Sahlmann, 2019-02-21:
Maybe the `view_image` example is already sufficient. I am using temporary files already for the table display, so maybe it is as simple as creating a second temporary files in csv format for example and point a variable like `jpg_filepath` to it. I will give it a try.

PR #415 (2019-05-20): Add download buttons to EDB page 
This PR introduces functionality in the web app code that allows a user to download the results of EDB mnemonic queries.

Closes #302
closed 2019-05-20 by bourque