spacetelescope/jwql #308: JWQL web app home page example query error

Johannes-Sahlmann, 2019-02-27:
When submitting the example program number `86600` in the search field of the web app home page, an error is raised: 
[image]

bourque, 2019-02-27:
Thanks for opening an issue @Johannes-Sahlmann.  I suspect this has to do with the recent switch to an updated filesystem that I recently made (#292), and likely this will introduce some issues in various places.

bourque, 2019-02-28:
FYI, I switched the path to the `filesystem` back to what it was before #292.  Until we can figure out some of the issues associated with the updated filesystem, I will leave it there to avoid errors with the web app.

bourque, 2019-03-04:
Now that #315 is merged, the original error that @Johannes-Sahlmann appears to be solved, however there is a new error:

[image]

laurenmarietta, 2019-03-05:
Weird stuff. I'll take a look!

PR #323 (2019-03-06): Always output instrument from filename_parser
Closes #308.

Turns out, the way that I was originally parsing the instrument out of a filename in `forms.py` only worked for certain kinds of filenames. After all of our updates to `filename_parser`, the instrument might appear in a filename in a variety of formats.

This PR adds another key to the output of `filename_parser` so that an `instrument` key is always returned.
closed 2019-03-06 by laurenmarietta