spacetelescope/jwql #309: Update generate_preview_images to handle different filename types

bourque, 2019-02-27:
While testing `generate_preview_images.py` on the updated filesystem (#292), I notice that it chokes on certain filename types, e.g. `jw00307-o003_t001_niriss_f277w-clearp_i2d.fits` results in

```
02/27/2019 13:25:58 PM CRITICAL: Traceback (most recent call last):
  File "/home/jwqladm/repositories/jwql/jwql/utils/logging_functions.py", line 238, in wrapped
    func(*a, **kw)
  File "/home/jwqladm/repositories/jwql/jwql/utils/logging_functions.py", line 203, in wrapped
    func(*a, **kw)
  File "generate_preview_images.py", line 512, in generate_preview_images
    grouped_filenames = group_filenames(filenames)
  File "generate_preview_images.py", line 626, in group_filenames
    observation = filename_parts['observation']
KeyError: 'observation'
```

This is because this particular filename type does not have an `observation` component to it.

bourque, 2019-02-27:
@bhilbert4 Do you know if `preview_image` supports generating preview images for `i2d` files?  Perhaps we should identify supported filetypes for `generate_preview_images` and restrict the software to only run on those.

bhilbert4, 2019-02-27:
This looks to me like a possible problem with the filename_parser. That filename you mentioned is not in the standard format.

bourque, 2019-02-27:
@bhilbert4 Well, it is technically the standard format for a stage 3 product, and we have actually updated the `filename_parser` to handle these (#248), but are just now realizing that `generate_preview_images` don't support these.

I think we should be able to update `generate_preview_images.py` to handle these, but I am just wondering if `preview_images.py` would have to be heavily refactored to generate preview images for `i2d` filetypes, and if so, a larger question of if we even want to support that should probably be asked.

bhilbert4, 2019-02-27:
`preview_images` should be set up to support 2, 3, or 4-D data, so I think it should be able to do an i2d file once it opens the file.

laurenmarietta, 2019-03-05:
Do we think this bug with `generate_preview_images` is responsible for all the images that aren't currently showing up on the web app? 

For instance: https://dljwql.stsci.edu/FGS/archive/00611/

bourque, 2019-03-05:
Hmmm, i'm not sure.  Let me dig into it.

bourque, 2019-03-05:
After some digging, I think the error @laurenmarietta points out is because the updated filesystem contains some programs that the old filesystem did not (`00611` being one of them) and so preview images have yet to be created for those.

I suspect that once we solve this issue (#309) and generate updated preview images, this problem will go away.

bourque, 2019-03-29:
@bhilbert4 I might need your help with implementing changes to `generate_preview_images` to support stage 3 filenames.  It appears that it is not going to be that straightforward.  For instance, some of the code is dependent on the existence of the `detector` in the filenames, but stage 3 filenames do not have a `detector` specified.  We will have to figure out how we want to group such files and make them available in the `view_image` webpages.  

Am I correct in understating that a stage 3 data product may be a product of several individual observations/rootnames?  Or is there still 1-to-1 correspondence?

bhilbert4, 2019-03-29:
Yes, I think you're right on that. Multiple observations can be combined in stage 3. I think in that case the default is to have the association name in some form within the filename.

bhilbert4, 2019-03-29:
Hopefully there are enough details in the pipeline/DMS documentation that we can figure out all the options.

bhilbert4, 2019-03-29:
One option may be to grab all the association files (json format) produced by DMS and examine them. Within an association file there is a field called `name` that contains the output name (minus suffix). Then we can search for files with that base name.

For example:
```
   "name": "jw96090-o001_t001_nircam_f322w2-f323n"
```

https://jwst-docs.stsci.edu/display/JDAT/Understanding+Associations

PR #357 (2019-04-11): Support stage 3 preview images and thumbnails
This PR modifies `preview_image.py` and `generate_preview_images.py` to support stage 3 data products.

Closes #309
closed 2019-04-11 by bourque