spacetelescope/jwql #311: Determine what to do with non-compliant filenames

bourque, 2019-02-28:
With the updated filesystem (#292), I ran `test_utils.test_filename_parser_whole_filesystem` and it kicked back 1578 files that could not be parsed, some common examples include:

```
jw00799003001_gs-acq1_2019154181705_stream.fits
jw00799003001_gs-fg_2019154183426_cal.fits
jw10005-c1001_t001_nircam_f430m-maskrnd-sub320a430r_psfstack.fits
jw10005003001_02102_00001_nrcalong_c1001_psfalign.fits
jw90001001002_02101_00001_nis_o001_crf.fits
w00729011001_gs-id_1_image_cal.fits
jw00729011001_gs-id_1_stacked_cal.fits
```

We should reach out to DMS folks and see what the dealio is with these types of files.

Johannes-Sahlmann, 2019-02-28:
*_gs_acq1_* are FGS guide star acquisition data
*_gs_fg_* are FGS fine guide data
*_gs_id_* are FGS identification step data

bourque, 2019-02-28:
@Johannes-Sahlmann Do you happen to know if those files are indeed in the proper naming convention and will permanently exist in the JWST archive?

Johannes-Sahlmann, 2019-02-28:
I am confident that they will exist permanently in the archive.
I do not know whether these examples are properly named.

bourque, 2019-02-28:
Ok, thanks!

laurenmarietta, 2019-02-28:
@shanosborne and I came across some FGS files like this in the filesystem the other day (the `gs-id`, `gs-acq`, and `gs-fg` files). They are products of the GUIDER pipeline and, like Johannes said, we should definitely expect to see files like this in the archive. 

I can ping some FGS people to see if I can determine what the naming structure for them will be.

I'm also happy to ask about the other filetypes, @bourque, if you haven't already?

bourque, 2019-02-28:
@laurenmarietta Sounds good, go for it!

PR #315 (2019-03-01): Fix filename parser for new filenames
Closes #311.

This PR brings `utils.filename_parser` up to snuff with the latest version of the DMS product filesystem.

In order to account for the weird guider files, I was able to get ahold of a TR (JWST-STScI-004800, SM-12) that defines the GUIDER data format:
[image]

Then, I found a [JDox entry](https://jwst-docs.stsci.edu/display/JDAT/Understanding+Associations?q=association%20candidate) explaining "Stage 2c" outlier detection associations products that have names of the form:
`jw<PPPPP><OOO><VVV>_<GGSAA>_<EEEEE>_<detector >_<AC_ID>_<suffix> .fits`
 [... cut]
closed 2019-03-01 by bourque