spacetelescope/jwql #312: Build function to hide exclusive access content in web app

bourque, 2019-02-28:
Before we build an external facing production web server (#294), we should build in a convenient and secure way to hide content within the web app if the user is not authenticated.

laurenmarietta, 2019-02-28:
When you say "content", I take it you mean specific material on a page rather than a whole page? FYI, maybe someone could see the EDB page but then not be able to submit a query, or something similar?

bourque, 2019-02-28:
@laurenmarietta Yeah, that is mainly what I meant, but you raise a good point.  There may be cases where we want to hide an entire page, so we will also have to address that functionality.  Perhaps we could do something like "if user isnt authenticated, redirect to a "not authorized" page".

bourque, 2019-03-04:
I am finding that there needs not exist a separate function for this, but rather just some conditions within `views.py` and/or specific HTML templates in the places where we want to hide the content.

PR #322 (2019-04-15): Hide certain content if user is not authenticated
This PR contains modifications necessary to hide specific areas of the web app (namely the viewing of preview images and thumbnails) if the user is not authenticated.

Closes #312 
Pertains to #294
closed 2019-04-15 by laurenmarietta