spacetelescope/jwql #325: Build diagram showing JWQL technologies and components

bourque, 2019-03-05:
It would be nice to have a general diagram that shows the overall technological landscape of the `jwql` project and how the various technologies/components relate to each other.

bourque, 2019-03-05:
Here is a rough draft:

[image]

@spacetelescope/jwql Any thoughts on how I might make this look better?  Can you think of anything I am missing? 

Attached is the file in case anyone would like to make edits. 
[jwql_technologies.pptx](https://github.com/spacetelescope/jwql/files/2932594/jwql_technologies.pptx)

laurenmarietta, 2019-03-05:
This is exactly the kind of procrastination I was looking for today 😈 

I think this might be too busy, but here's another option (including a new "project management" section and moving things around a bit):
[image]

Apparently you can't attach a keynote file (easier to make logos transparent than in PPT) to GitHub, but I'm happy to Slack it to anyone who wants it.

bourque, 2019-03-06:
Thanks @laurenmarietta, this is awesome!! Good call on including Jenkins, I totally forgot about that.  

I wonder if it would be better to change "Software Management" to "Software Tools" instead, and add logos for `pytest` and `bokeh`.  And then in "Project Management" add logos for `Anaconda` and `ReadTheDocs`.  What do you think?

I'd be happy to play around with it if you want to send me the file.

bourque, 2019-03-06:
Here is a slightly updated version:

[image]

bourque, 2019-03-06:
I will close this issue for now.  We can make tweaks to this as we see fit.

closed 2019-03-06 by bourque