spacetelescope/jwql #341: Data-Trending - ToDo's

DanielKuebi, 2019-03-21:
Data Trending for NIRSpec and MIRI works fine in each seperate branch. To merge it to the master from my side following should be done: 

**Must haves**

-> Implement cron job (MIRI and NIRSpec)
During development the EDB was not filled with real test data. We queried 15min samples for each day and whole-day samples with the WOW tool and had the data in seperate .CSV files. Therefore `/utils` holds `csv_to_AstropyTable.py` which is used for importing the data from the CSV files and create a dict specified below [1]. To run the program daily it should be able to query the data from the EDB and process it with the 'process_file' function implemented in `day_to_db.py` and `15min_to_db.py`. I have already copied them into `dt_cron_job.py` and prepared them for future use. The *_to_db.py files will no longer be required in future. 

-> Add slider or boxes to set date range (MIRI and NIRSpec) 
In `dashboard.py` you can choose the start and end time for the plots. Maybe they should show just the last 4 months by default but it is important to be able  to change the date-range and update the plots. Therefore a date-range slider or two imput boxes for start and end time should be implemented. Their values cann be passed to all plots in `dashboard.py`. 

-> Commenting functionality/ Jira plugin (MIRI and NIRSpec) 
If anomalies or a trend occurs it is important to note that down. Therefore a convenient commenting functionality must be implemented. 

-> Hardcoded descriptions (MIRI and NIRSpec) 
Every tab in the dashboard starts with a short description. These descriptions are currently hardcoded and added to the *tab.py files in `/plot` folder. It is more convenient to create seperate files for the descriptive tables and import them in the *tab.py files. Besides seperating the information it would also be nice to maybe import information from the EDB and show it in these tables. I have not yet figured out a convenient way to do so... 

**Nice to have**

-> Preparing plots once a day
I think the rendering can get pretty slow when we have a lot of data. Therefore it would be useful to call `dashboard.py` once a day and store the returned bokeh components temporarly. Whenever someone want´s to look at the monitor we can supply the information much faster and wont have a high latency due to querying the auxiliary database and preparing the plots. 

-> Performance check 
If possible please check the performance of the programm with more datapoints than we have at the moment. It would be bad, if there are errors during the mission because the dashboard cannot handle such a big dataset. I unfortunately had no way to test the stability with a large dataset.

bourque, 2019-03-21:
@DanielKuebi Thanks for making this, it will be quite helpful for us!

bourque, 2020-03-26:
This is partially addressed by #547

PR #832 (2022-11-11): [WIP]: Add EDB telemetry monitor
This PR adds the refactored engineering database telemetry monitor.

Starting with the code as originally written, and using the documentation for the MIRI monitor as a guide, we plan to adapt the EDB telemetry monitor to make it useful for all instruments.

Still early days here. Lots of modifications still to make.

Resolves #814 
Resolves #351 
Resolves #341
closed 2022-11-11 by mfixstsci