spacetelescope/jwql #355: Make MIRI and NIRSpec DB just availlable on the STScI Server.

DanielKuebi, 2019-04-02:
Since all the code of the MIRI and NIRSpec date trending monitor is now merged I would strongly consider to make the auxiliary databases for the monitor availlable just on your local filesystem. Although we don´t have raw data online, the NIRSpec DB contains still some ITAR restricted mnemonics (ASIC Voltages).

DanielKuebi, 2019-04-02:
@bourque

bourque, 2019-04-02:
Thanks @DanielKuebi, yes, I think the plan is to ingest these databases into internal postgesql servers hosted on internal VMs (as we do with our other database tables) -- in that case, only users with the proper credentials will be able to access it.  This will likely be taken care of when we address #351.

closed 2019-04-02 by bourque