spacetelescope/jwql #359: Create JIRA workflow/support structure for contributing instrument monitors

bourque, 2019-04-08:
I think we could use Jira workflows to make it easier for contributors (especially those external to the `jwql` dev team) to contribute implementable instrument monitors.  The workflow should make apparent which steps need to be undertaken in order to contribute a working instrument monitor to the `jwql` package, from the initial steps of determining requirements and gathering existing materials, to the final steps of opening a pull request.

Johannes-Sahlmann, 2019-06-05:
We now have a `Monitor` issue type and an associated workflow that we can customize: https://jira.stsci.edu/plugins/servlet/project-config/JWQL/workflow/?name=Monitor%20Development

bourque, 2019-07-01:
@Johannes-Sahlmann @cracraft @laurenmarietta @bourque have met and come up with a draft workflow.  We will add a `README` to the JWQL Github repo that describes this workflow to contributors. 

For the time being, here is the current landing page for this: https://innerspace.stsci.edu/display/JWQLPROJ/Development+of+monitoring+scripts+by+the+JWQL+project+team

bhilbert4, 2024-05-13:
JWQL is now managed through Jira, and includes a workflow that instrument developers can use.