spacetelescope/jwql #363: Make test xfails into skips

laurenmarietta, 2019-04-15:
We have a lot of `pytest.mark.xfail` tests in our suite that I think should be converted into skips -- like @bourque added in #356: https://github.com/spacetelescope/jwql/blob/f3d4a3303fa2c786df21937c8cf7831c3e63b20c/jwql/tests/test_preview_image.py#L56

In my opinion, it is more accurate to skip tests on the Jenkins server if they can't be run there, rather than marking them as `xfail`. I think some of those tests really should be passing on our local machines, but marking them `xfail` makes it hard to tell if something has actually broken.

laurenmarietta, 2019-04-15:
(As I'm getting caught up, I see that #356 tackled part of this, but I think `test_api_views` and `test_edb_interface` can be altered as well.)

bourque, 2019-04-15:
@laurenmarietta While we are at it, we should probably also mark the tests in `test_loading_times` with `skipif` as well, since those require access to the `config.json` file:

https://ssbjenkins.stsci.edu/blue/organizations/jenkins/STScI%2Fjwql/detail/PR-322/11/pipeline/21/#step-39-log-21

PR #397 (2019-05-20): Skip tests
Closes #363.

This PR:
- Rewrites tests in our test suite to use `skip` instead of `xfail`, so that the test suite is more accurate when run both on Jenkins and locally
- Updates `test_api_views.py` so that the API tests are run on the dev server during Jenkins builds
- Renames one of the file roots in `test_api_views` that was removed from the filesystem

Note that there is a temporary `if` statement in `test_api_views` that skips any test that throws the "502 Bad Gateway" error (#407) . This should be removed in the future.
closed 2019-05-20 by bourque