spacetelescope/jwql #364: Jenkins builds still passing despite numerous failed unit tests

bourque, 2019-04-15:
It seems that Jenkins builds are passing even though there may be several failed unit tests, for example:

https://ssbjenkins.stsci.edu/blue/organizations/jenkins/STScI%2Fjwql/detail/PR-322/11/pipeline/21

#338 attempted to fix this, but it doesn't seem to be having an effect afterall.

laurenmarietta, 2019-04-15:
Is it possible that the Jenkinsfile needs to explicitly state
```
bc0.failedUnstableThresh = 0
bc0.failedFailureThresh = 0
```
rather than removing the lines and not defining them at all?

bourque, 2019-04-15:
@laurenmarietta Hmm, good idea, that could be it

obviousrebel, 2019-04-16:
@bourque Just taking a quick look.
`https://ssbjenkins.stsci.edu/blue/organizations/jenkins/STScI%2Fjwql/detail/PR-322/11/pipeline/21/#step-39-log-21`

Seems that build log is no longer available. Looking at one of the latest builds, perhaps:
```
No .xml files found in workspace. Test report ingestion skipped.
```
is causing this? Seems like Jenkins isn't ingesting the test report and therefore doesn't know about the failures?

rendinam, 2019-04-16:
@obviousrebel is correct. The multiple .xml files in the repository root after testing and coverage measurement are performed are causing the problem. This is a known limitation. The jenkins_shared_ci_utils docs state:

"After the last test_cmds command is executed, Jenkins examines the build environment for a filename with an .xml extension. If one is found, it is assumed to be a JUnit-compliant test report and is read. "

If the coverage reporter can be told to expect the report in a location other than the repository root, that would be the quickest temporary fix. The CI utils library will ultimately have to receive some logic to be more discriminating of which files represent test reports and are thus ingestible by Jenkins. https://github.com/spacetelescope/jenkins_shared_ci_utils/issues/45

bourque, 2019-04-19:
Thanks @rendinam and @obviousrebel, I think I've got this working; the Jenkins build is now finding only one report and failing expectedly (one of the few times I am happy to have failing tests around!):  https://ssbjenkins.stsci.edu/blue/organizations/jenkins/STScI%2Fjwql/detail/PR-373/7/pipeline/

Here is the solution I came up with:

```
bc0 = new BuildConfig()
bc0.nodetype = "linux-stable"
bc0.name = "debug"
bc0.build_cmds = [
    "conda env update --file=environment.yml",
    "pip install codecov pytest-cov",
    "with_env -n jwql python setup.py install"]
bc0.test_cmds = [
    "with_env -n jwql pytest -s --junitxml=results.xml --cov=./jwql/ --cov-report=xml:coverage.xml",
    "codecov --token=${codecov_token}",
    "mkdir -v reports",
    "mv -v coverage.xml reports/coverage.xml"]
```

PR #373 (2019-04-26): Fix passing Jenkins builds despite failed unit tests
This PR attempts to configure the `Jenkinsfile` such that Jenkins builds will fail upon failed unit tests.

Closes #364
closed 2019-04-26 by bourque