spacetelescope/jwql #374: Instrument-specific dark monitor updates

bhilbert4, 2019-04-19:
#229 adds the Dark Current Monitor to JWQL. The scope of the added functionality in that PR is to get the dark current monitor working for NIRCam. Some additional tweaks will be necessary in order to make the code functional for all of the other instruments. 

The purpose of this issue is to gather together the edits needed for the other instruments.

Known updates needed so far:

### MIRI Amplifier Definitions
The amplifiers on MIRI read out pixels in a different pattern than in any of the other instruments. For the other instruments, each amplifier reads out a contiguous block of pixels (e.g. from (x,y) = (0,0) to (511, 2047)). But in MIRI the amplifiers read out interleaved columns (double check that it is columns rather than rows) of pixels. (i.e. Columns 0 - 7 are read out by: amp1, amp2, amp3, amp4, amp1, amp2, amp3, amp4).

The amplifier definitions currently in the `amplifier_info` function cannot handle this pattern at the moment. It should be updated to do so, or possibly fall back to a situation where we say that all pixels are read out by a single amplifier. Could be interesting to pull apart and monitor the pixels from each amp separately, even though they are interleaved.

### MIRI Subarrays
All MIRI subarrays are read out using all 4 amplifiers. This affects the frame time calculation. Be sure this information is not needed elsewhere.

### MIRI Frametime Calculation
@mwregan2 sent me a notebook with a function that calculates the MIRI frame time for a given aperture. This needs to be added to the frame time calculation function.

### NIRSpec Amplifier Definitions
Request sent to @cherylpavlovsky asking about which pixels are read out by which amplifiers

### NIRSpec Frametime Calculation
Request sent to @cherylpavlovsky asking about frame time calculation.

### NIRSpec Subarrays
Request sent to @cherylpavlovsky asking about number of amplifiers used to read out various subarrays

### NIRISS Subarrays
All NIRISS subarrays are read out using a single amplifier, except for four: WFSS64C, WFSS64R, WFSS128C, and WFSS128R. The latter two should be added to `FOUR_AMP_SUBARRAYS` in `utils/constants.py`. The first two are already present there.

bourque, 2019-04-20:
I noticed a possible issue with the database entries associated with the dark monitor: The `x_coord` and `y_coord` columns of the `nircam_dark_pixel_stats` table are blank with entries of `{}`

Additionally, we will still need to figure out if and how we want to populate the `affected_tables` and `status` fields of the `monitor` table; this will require a bit of a refactor to the current code that populates this table, so I decided to hold off on that and address it in a future PR (perhaps the one that comes out of this issue).

bhilbert4, 2019-04-22:
The empty bad pixel lists are ok. If there is no previous mean slope image to compare to, then no bad pixels will be found. On the next run of the dark monitor, with a new set of data, the search should be performed.

bourque, 2019-04-22:
@bhilbert4 I have made a branch and WIP PR for this, see #377.  I've included some updates to address my concerns about the `monitor` table.

PR #377 (2019-08-07): dark_monitor for all instruments
This PR makes appropriate changes to the `dark_monitor` to get it working for all JWST instruments. 

Closes #374
closed 2019-08-07 by bourque