spacetelescope/jwql #386: Update dashboard plot axis labels

laurenmarietta, 2019-04-24:
Now that the filesystem monitor has been updated (#328), it looks like the axis labels for the different dashboard (https://dljwql.stsci.edu/dashboard/) plots should be updated. I see at least a couple that seem to have the y axis labeled as "counts" when in reality the plot is showing some multiple of bytes.

cracraft, 2019-04-26:
I was looking at the pages, and all of the file size by instrument plots have Counts listed where it should be size (TB) listed. Only the System stats plot has a size indicator, and that's listed as GB where I think it should be TB. I have some other cosmetic changes that I can either add to this ticket or open a new issue. The plots don't currently seem organized well. The top of the page shows MIRI File counts next to FGS file size and the plots are randomly shown in no particular order. Also, bokeh should allow us to put in the tools like pan, zoom, etc., but those don't show up on the filesystem monitor plots.

cracraft, 2019-04-26:
Also, the color palette used is Dark2_5, which cycles between only 5 colors. There are up to 20 distinct file types in the 'total file counts by type' plot, which has a lot of indistinguishable color repeats. We should at least be using something like Category20_20 that has 20 different colors for fewer repeats.

cracraft, 2019-04-26:
I realize that the bokeh tools issue and the arrangement of the monitor plots are part of the webpage/plot interface rather than in the code of the monitor_filesystem, so I'll open a separate issue for that.

bourque, 2019-04-29:
Closed via #390.

closed 2019-04-29 by bourque