spacetelescope/jwql #388: Create base class for instrument monitoring

bourque, 2019-04-26:
@jwhite3 gave a presentation regarding the COS monitoring tool in which he showed a framework for building instrument monitoring software; the framework consists of a base `monitor` class that contains attributes and methods common to all monitors.  This base class then gets inherited by specific instrument monitoring classes. 

I wonder if JWQL could also benefit from a framework like this.  The `base` class could contain attributes and methods for performing common monitoring tasks.

jwhite3, 2019-04-29:
@bourque I should have the code up on GitHub today, so once that's done I can post the link here

bourque, 2019-04-29:
@jwhite3 Awesome, thanks!

jwhite3, 2019-05-01:
@bourque https://github.com/spacetelescope/monitor-framework

jwhite3, 2019-05-01:
Beware though, If you're thinking about using it directly, it's kind of under-construction at the moment. 

However, if you're just interested in inspiration, dive in!

bhilbert4, 2021-10-01:
@bourque it seems like [monitor-framework](https://github.com/spacetelescope/monitor-framework) would be a good place to start. No work on it in the last year and a half. Looks like James has left ST. Maybe we should see if anyone is using this and if not, use it ourselves? Or make a copy and start from there?

stscijgbot-jwql, 2024-05-13:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-236?focusedCommentId=698811#comment-698811):_

This could certainly help for future monitor development, although it's hard to say how much. Some of the basics (db interactions, finding data in the filesystem, etc) are probably common to a lot of monitors.