spacetelescope/jwql #391: Organize filesystem plots on Dashboard

cracraft, 2019-04-26:
When I look at the Dashboard, the plots don't currently seem organized well. The top of the page shows MIRI File counts next to FGS file size and the plots are randomly shown in no particular order. Also, bokeh should allow us to put in the tools like pan, zoom, etc., but those don't show up on the filesystem monitor plots. I think these issues have to be fixed on the website side of things. Is there a way to set a specific order? Look at filesystem_monitor.html for an example. Total plots should come first, then instrument plots, with each row of plots belonging to a single instrument. (I first mentioned this in #386, but that Issue dealt just with the code for the filesystem monitor plot creation.

bourque, 2019-04-29:
@cracraft Yes, I think you are right in that logic will have to be employed on the webapp side (i.e. probably `data_containers.get_dashboard_components`) to enforce an order to the plots.  Currently, this function simply walks through the `ouptuts` directory to gather materials to render in a non-ordered way.

bourque, 2020-10-20:
This will be solved with #621

closed 2021-04-01 by bourque