spacetelescope/jwql #392: EDB query times incorrect?

laurenmarietta, 2019-04-26:
I've been interacting a lot with the EDB web page today, and I keep seeing some weird behavior where it looks like the EDB query results plot is completely ignoring the start and end date/time:
[image]
@Johannes-Sahlmann have you seen this before?

Johannes-Sahlmann, 2019-04-29:
No I have not seen this before, but it may be a corner case of the MAST service that we use. 
As a double check, can you reproduce this situation with a programmatic query (not via the web app)?

I could look into this as part of the EDB wrapup in the current sprint.

laurenmarietta, 2019-04-29:
The questionable time problem does unfortunately seem to be occurring in the programmatic query as well. Including here a screenshot of trying to query two days in September 2017 and getting back two entries: one from February 2017 and another from August 2018:

[image]

Johannes-Sahlmann, 2019-05-08:
I can reproduce this behaviour (approximately) using the web portal (https://mast.stsci.edu/portal/Mashup/Clients/JwstEdb/JwstEdb.html) so this is likely not an issue of `jwql` itself.
Strangely, the web portal query returns three entries (see attachment) whereas the programmatic queries using jwql return two entries. I will forward this to MAST. 

[image]

[IGDP_MIR_IC_DET_TEMP-20170214T230000-20180808T184434.txt](https://github.com/spacetelescope/jwql/files/3158680/IGDP_MIR_IC_DET_TEMP-20170214T230000-20180808T184434.txt)

Johannes-Sahlmann, 2019-05-08:
According to MAST staff, this is the expected behaviour: 
"The EDB returns from the datapoint that precedes the requested start time to the datapoint that follows the requested end time, thus ensuring a containment. "

We should possibly include a note on this in the documentation of jwql and/or jwedb

Johannes-Sahlmann, 2019-05-10:
I added some notes on this in the relevant docstrings, see #402

closed 2019-05-10 by Johannes-Sahlmann

laurenmarietta, 2019-05-12:
Thank you for figuring this out, @Johannes-Sahlmann!
I wonder whether the EDB plotting feature on the web app should be altered, now, so that only the points in the requested timeframe (if there are any) are shown. Probably a discussion for a separate issue.