spacetelescope/jwql #398: test_get_pipeline_steps failing on 3.5 Jenkins

laurenmarietta, 2019-05-03:
Seems as though the new Python 3.5 Jenkins build added in #387 doesn't quite succeed.

The assertion in `test_get_pipeline_steps()` in `test_pipeline_tools.py` is failing:
https://ssbjenkins.stsci.edu/blue/organizations/jenkins/STScI%2Fjwql/detail/develop/17/pipeline/22

I assume (hope!) this is a simple version issue.

(Also, why isn't this failing the Jenkins build??? The question that never goes away...)

obviousrebel, 2019-05-07:
The failing test I believe is not failing the build because Jenkins is not ingesting the results.xml file generated by pytest as it states it is invalid junit format.

In order for pytest to create junit valid xml reports, a pytest.ini file must be present to specify the junit family (which is present in this repo). Unfortunately, pytest added this feature in 4.2 and the latest available version of pytest for python 3.5 is 3.8.1.

Jenkins is complaining that
```
Attribute 'line' is not allowed to appear in element 'testcase'.
```

Which is what makes the result.xml file invalid (to jenkins). One 'hacky' way around this would be a script or sed command that runs on only the python 3.5 build to regex match and remove that attribute (line=*) from all test cases in the results.xml file to appease Jenkins and allow processing of the test results.

bourque, 2019-05-14:
#412 will hopefully solve the false Jenkins reporting, but we will still need to figure out why `test_get_pipeline_steps` is failing (e.g. https://ssbjenkins.stsci.edu/blue/organizations/jenkins/STScI%2Fjwql/detail/PR-410/3/pipeline/22#step-147-log-34)

bourque, 2019-05-14:
I have been able to reporoduce the failing test locally, and I believe the issue is that for some reason python 3.5 stores the `req_steps` in a different order than the `steps_dict`, and since the `assert` statement is comparing two `OrderedDict`s, the `assert` fails.  For example:

```
        print(instrument)
        print(req_steps)
        print(steps_dict)
```

returns:

```
OrderedDict([('dq_init', True), ('saturation', True), ('superbias', True), ('refpix', True), ('linearity', True), ('persistence', True), ('dark_current', True), ('jump', True), ('rate', True), ('firstframe', False), ('ipc', False), ('lastframe', False), ('group_scale', False), ('rscd', False)])
OrderedDict([('dq_init', True), ('saturation', True), ('superbias', True), ('refpix', True), ('linearity', True), ('persistence', True), ('dark_current', True), ('jump', True), ('rate', True), ('group_scale', False), ('ipc', False), ('firstframe', False), ('lastframe', False), ('rscd', False)])
```

I think this could be mitigated by converting the `OrderedDict` to a normal `dict` type, and then comparing.  @bhilbert4 Is it absolutely necessary that the contents of these dictionaries be in a specific order?  Or can we bypass that for testing purposes?

bhilbert4, 2019-05-14:
The point of the test is to make sure that the proper pipeline steps needed for a given file are returned in the order that the pipeline runs them. So we could convert the `OrderedDict` into a normal `dict` and compare, although that would essentially be removing half the point of the test to begin with. But at that point, I guess we'd still be testing that the correct steps are being returned. Maybe that's enough given that the alternative is a failing test.

laurenmarietta, 2019-05-14:
It looks to me like the discrepancy is coming from how pipeline steps that _don't_ need to be run are added to the `req_steps` OrderedDict. Steps that won't be run are pulled using a for loop from the dictionary`pipeline_tools.PIPE_KEYWORDS`. Maybe the way Python 3.5 iterates through the values in the `PIPE_KEYWORDS` dict is different than the order that Python 3.6 iterates through them?

@bhilbert4, would it be correct to say that the order for the steps that the pipeline is _going_ to run (i.e. those marked `True` in the dicts) is important, but the order of the steps that the pipeline is _not going_ to run (i.e. those marked `False` in the dicts) is not important? If so, I imagine we could re-write the test to reflect that.

bhilbert4, 2019-05-14:
Weird. Yeah, we could conceivably get rid of all the False entries and just compare the dictionaries of required steps.

bourque, 2019-05-14:
Thanks for the suggestion @laurenmarietta, I modified the tests to only test the (ordered) steps that need to be run and updated #414

PR #414 (2019-05-14): Fix test_get_pipeline_steps for python 3.5
This PR modifies `test_pipeline_tools` in order to get the tests within to pass in a Python 3.5 environment.

Closes #398
closed 2019-05-14 by bourque