spacetelescope/jwql #406: Unable to login when running web app locally

Johannes-Sahlmann, 2019-05-10:
I want to log into the web app when running it locally to develop code related to MAST authentication. When I click the `Login` button on the top right of the page, I get a 'Server not found' (We can’t connect to the server at oauth.) error.

When I do the same on our development server, I get to the page where I enter my AD credentials. After doing that I get a different error:
```
NoReverseMatch at /authorize/
Reverse for '' not found. '' is not a valid view function or pattern name.
```

Do other users have the same problem?

Maybe something is missing in my config file? If so, the application should raise an informative error.

laurenmarietta, 2019-05-12:
Very strange... I have run into that second error before, but only ever on a feature branch. I cannot reproduce what you are seeing on `develop` or `master`.

I would say the "auth_mast", "client_id", "client_secret", and "mast_token"  keys in your config file are the important ones for this? I assume yours are correct, though, so I doubt those are causing the problem.

I am very curious if anyone else has run into this, and in what contexts.

Johannes-Sahlmann, 2019-05-13:
I think it must be related to my config file. I had `auth_mast` pointing to an empty string. Once I put in the correct address I am getting one step further. 
now I need to check for `client_id` and `client_secret`

It would be great to catch such problems and return an informative error. I am not sure where these checks should be done, though.

Johannes-Sahlmann, 2019-05-13:
After setting `client_id` and `client_secret` I can log in when running locally.

Johannes-Sahlmann, 2019-05-13:
I am however still unable to log in when using the development server.

laurenmarietta, 2019-05-13:
Hmm... Do you also have your MAST token set as an environment variable (`$MAST_API_TOKEN`)? As described here: https://auth.mast.stsci.edu/info I'm not _sure_ if the dev server web app uses that, but I imagine it might.

I 100% agree about having a more informative error message. The errors that you were experiencing seemed pretty specific; maybe we can add a try/except clause and raise a more informative error if, say, that "no such table: django_session" message comes up. I'll make a new issue for that.

Johannes-Sahlmann, 2019-05-13:
thinking about it a bit more: Does it make sense that a user is required to have a mast token stored in a local environment variable in order to log into the jwql web app running on the server?
It seems to me that this defeats the purpose of using auth.mast

bourque, 2019-05-14:
I think the error:

```
NoReverseMatch at /authorize/
Reverse for '' not found. '' is not a valid view function or pattern name.
```

might have to do with changes related to #334, I can look into this.

bourque, 2019-05-14:
@Johannes-Sahlmann Good point about requiring a user to store a token locally, though I'm not sure of a way around this.

bourque, 2019-05-30:
With #422 and #417 merged, I will consider this closed.

closed 2019-05-30 by bourque