spacetelescope/jwql #407: 502 Bad Gateway for dev server API views

laurenmarietta, 2019-05-12:
I first noted this in #397, but the problem hasn't gone away.

For certain API views on the web app via the dev server, e.g. https://dljwql.stsci.edu/api/nircam/thumbnails/, the web app is returning a "502 Bad Gateway" error. (See screenshot below.)

Interestingly, these views still render for me when I run the web app locally - but they take a while to load. I wonder whether there is some sort of timeout happening?

[image]

bourque, 2019-05-17:
@laurenmarietta I suspect you are correct about the timeout.  Looking at the web app logs when trying to access this page, I see this:

```
May 17 14:10:09 dljwql.stsci.edu gunicorn[10533]: [2019-05-17 14:10:09 -0400] [10533] [CRITICAL] WORKER TIMEOUT (pid:12654)
```

I can dig into if there is a setting that we could adjust, or perhaps Alex Yermolaev has some ideas.

bourque, 2019-05-17:
It looks like there are some settings within `gunicorn` and/or `nginx` that might solve this (https://stackoverflow.com/questions/10855197/gunicorn-worker-timeout-error, https://serverfault.com/questions/490101/how-to-resolve-the-gunicorn-critical-worker-timeout-error), however we will need to ask Alex to make the appropriate changes in the config files since we don't have write permission.

laurenmarietta, 2019-06-05:
(Once this is fixed, we should remember to un-skip the tests that I marked to skip because of this error.)

bourque, 2019-06-12:
I've reached out to Alex to try to solve this issue

bourque, 2019-06-12:
Alex has increased the timeout amount and the API view appears to be rendering now!

closed 2019-06-12 by bourque