spacetelescope/jwql #411: Increase tolerance for test_gaussian1d_fit

laurenmarietta, 2019-05-14:
I've seen a few Jenkins builds fail here and there because `test_gaussian1d_fit()` fails sporadically. Example: https://ssbjenkins.stsci.edu/blue/organizations/jenkins/STScI%2Fjwql/detail/PR-378/6/pipeline/23/#step-54-log-34

My guess as to why it only happens sometimes is because of the randomness inherent in the test. I think if we increased the tolerance for the `np.isclose()` statement in that test, it wouldn't fail any builds. Though I'll let @bhilbert4 comment, since he wrote it!

bhilbert4, 2019-05-14:
I just ran the test ~20,000 times and I think I have a good idea now of the tolerance to use that should allow the test to pass. I think bumping it up from 0.0015 to 0.0030 should be enough. The largest difference I saw in my testing was 0.0024. I can add this into #377

bourque, 2019-05-14:
@bhilbert4 Since this is causing random build fails for other PRs (e.g. #410, #378), I suggest to make this adjustment now in a separate PR and get it into `develop`.  Would you be able to to that today or tomorrow?  If not, i'd be happy to take care of it!

PR #413 (2019-05-14): Relax requirements for gaussian 1d test
This PR relaxes the tolerances for the fitting results for the `test_gaussian1d_fit` functions. The previous tolerances were too tight and causing Jenkins to fail.

Closes #411.
closed 2019-05-14 by bourque