spacetelescope/jwql #425: EDB mnemonic query plot blank on local server

laurenmarietta, 2019-05-22:
@Johannes-Sahlmann and I have both noticed that, when the web app is being run on a local server, the EDB mnemonic query `bokeh` plot does not appear:
[image]
I personally see this behavior on every branch - feature branches, `develop`, and `master`.

Good news is, I don't see this bug on the development server (dljwql.stsci.edu). This suggests to me that it has something to do with running the local Django server? 

If anyone else has seen this, please share.

bourque, 2019-06-03:
@laurenmarietta @Johannes-Sahlmann Here is what the javascript console shows when trying to build a plot:

[image]

Johannes-Sahlmann, 2019-06-03:
Do you know how to fix this? Is it something that I need to add to the python bokeh-plot generation code? 
How can I display the javascript console?

bourque, 2019-06-03:
This appears to be an issue with the version of `bokeh`.  The dev web server is currently running version `1.0.1`, while our current `jwql` environment supports `1.1.0` (the env on the dev server has to be manually updated since it is installed in a very specific place, hence why it is out of date).  When i downgraded my local version to `1.0.1` I was able to render a plot with no error messages from the javascript console.

Reading the error message, it seems that we might be able to get this working in version `1.1.0` if we declare `Plot.center` somewhere in the code(?) (I'm not exactly sure what this means).

bourque, 2019-06-03:
@Johannes-Sahlmann Using chrome, you can go to View > Developer > JavaScript Console.

bourque, 2019-06-03:
@Johannes-Sahlmann I was able to solve this by simply updating this line to the newer version number:

https://github.com/spacetelescope/jwql/blob/30935d4230a51165d2f0bb7b4305c8d0c13991f4/jwql/website/apps/jwql/templates/base.html#L25

I will put the change in a PR (and make sure it is not breaking anything else)

PR #429 (2019-06-11): Update web app bokeh version
This PR updates the web app to use bokeh version `1.2.0`, which is the current supported version in the `jwql` environment.  This also fixes the issue described in #425 

Closes #425
closed 2019-06-11 by bourque