spacetelescope/jwql #426: Download EDB data from dev server

laurenmarietta, 2019-05-29:
The way that EDB data downloads were downloaded in #415 seems to not be working on the development server.

This might be a simple permissions error, which would hopefully be easy to fix....

It might also be a problem with where the files are saved - they might need to be put on central store instead of inside the dev server repo?

While this is fixed, it might also be worthwhile to update the code to save the files with unique names based on the query time, e.g. somewhere like jwql/temp_edb_files/edb_exploration_20190516_204343.csv. Then there could be a cron job that cleans out that directory every day, hour, five minutes, etc.

laurenmarietta, 2019-05-29:
Potentially elegant solution using Django to dynamically generate CSVs and return them with an HTTP response: https://docs.djangoproject.com/en/2.2/howto/outputting-csv/

Johannes-Sahlmann, 2019-12-03:
Unfortunately, I don't think I will be able to fulfill this assignment.