spacetelescope/jwql #434: Update jwql environments on dev web server

bourque, 2019-06-05:
Now that we have specific environments tied to python versions, we should update the deployed `jwql` environment on the dev web server that is running the dev web app to reflect these new changes.

bourque, 2019-06-21:
I have installed `jwql-3.6` env on the dev server, which the web app is now using.

closed 2019-06-21 by bourque

laurenmarietta, 2019-06-25:
Would it be helpful to add a cron job to update the environment according to changes made to the environment file every time it pulls from the github repo? Or maybe less frequently, but still automated?

bourque, 2019-06-25:
@laurenmarietta Good idea!  I can try getting this to work alongside the `cron` job that pulls `develop`, though since `conda` can be finicky and sometimes take a while, probably best to just do it perhaps once an hour instead of every five minutes.